frostney/react-native-ibeacon #17: Android support?

httpNick, 2015-09-16:
I was wondering if this module would work with Android? I see the example is for iOS.

frostney, 2015-09-17:
It's not available for Android yet. I'd happily accept pull requests that add this functionality though. :)

octavioturra, 2015-09-29:
Does it have any movement? Is someone working on it?

frostney, 2015-09-30:
Unfortunately there's no one working on that at the moment. I don't have access to my Estimote beacons for the next 6 weeks, as much as I would like to work on that myself.

joevo2, 2015-10-19:
+1 support for android would be great :'( wanted to use this for my final year project but i think i would need to resort to ionic instead of RN

frostney, 2015-12-08:
As far as I know Android support is not as easy as the iOS version, primarily because there is no official SDK for iBeacons for Android. You might want to try @octavioturra AltBeacon implementation (https://github.com/octavioturra/react-native-alt-beacon) which I'm going to link in the readme.

albo1337, 2016-03-21:
Is the AltBeacon android implementation of @octavioturra with estimote beacons working?