frostney/react-native-ibeacon #19: Not Receiving DidEnter and DidExit

bradynapier, 2015-10-26:
I am not currently receiving these events - am I doing something wrong?  Works just fine when I use ranging.

var region = {
    identifier: 'myid.beacon.transmitter',
    uuid: 'f7826da6-4fa2-4e98-8024-bc5b71e0893e'    
};

Beacons.requestWhenInUseAuthorization();
Beacons.startMonitoringForRegion(region);

var subRegion = DeviceEventEmitter.addListener(
  'regionDidEnter',
  (data) => {
    console.log('!!!!! ---- |||| ==== !!!! ---- !!!=== !!!---');
    console.log('!!!!! ---- |||| ==== !!!! ---- !!!=== !!!---');
    console.log('!!!!! ---- |||| ==== !!!! ---- !!!=== !!!---');
    
    console.log('Region Did Enter!');
    console.log(data);
    
    console.log('!!!!! ---- |||| ==== !!!! ---- !!!=== !!!---');
    console.log('!!!!! ---- |||| ==== !!!! ---- !!!=== !!!---');
    console.log('!!!!! ---- |||| ==== !!!! ---- !!!=== !!!---');

  }
);

tioback, 2016-06-25:
@bradynapier did you manage to solve this? I'm also stuck with monitoring events not being called.

leomar-developer, 2016-11-03:
Does anyone knows if this is still a bug or something else needs to be done in order for these callbacks to work? I'm having the exact same issue.

tioback, 2016-11-03:
It worked once, but I cannot reproduce it.

leomar-developer, 2016-11-03:
I can confirm it is NOT working in latest version (0.6.0), ranging beacons works fine though

leomar-developer, 2016-11-08:
Can someone please shed some light on this one? I would really appreciate if someone can help out, we are using react 0.35.0 and a device with iOS 9. Ranging works fine but region monitoring seems to not be working we have tried a lot of variations on the code and "regionDidEnter" never gets called

leomar-developer, 2016-11-08:
Just solved this one, I will describe what I did just in case it happens to be the same issue for the folks that also reported this above. The issue was that (for some strange reason) the popup dialog that asks the user to accept the usage of the background location was not coming up (even after properly adding the NSLocationAlwaysUsageDescription value on Info.plist), so I had to manually go to the app settings and tell it that the Location was needed "Always", after doing that and relaunching the app the events started to fire as expected.

tioback, 2016-11-11:
@leomar-developer, thanks!did you try forcing the permissions popup to show?

montehansen, 2016-12-05:
I have the same problem in that regionDidEnter and regionDidExit do not fire, actually, it's kind of flaky; sometimes the enter will fire sometimes not (mostly not). In my case, I verified that Location was set to Always and have also tried WhenInUse without changed. It seems that ranging works, and monitoring doesn't. I am using Estimote iBeacons.

react-native-cli: 1.2.0
react-native: 0.38.0
react-native-ibeacon: ^0.6.0