frostney/react-native-ibeacon #20: Update readme / AppStore rejection

grabbou, 2015-11-24:
Hey, we've submitted the app and here's what we got:

> 2.16 Details

> Your app declares support for bluetooth-central in the UIBackgroundModes key in your Info.plist, but does not provide Bluetooth Low Energy functionality. The Bluetooth Background modes are for applications that communicate to other devices using Bluetooth Low Energy and the Core Bluetooth framework.

> Next Steps

> Please revise your app to add support for Bluetooth Low Energy communication through Core Bluetooth in your application or remove the bluetooth- values from the UIBackgroundModes key.

After googling for a while and checking source code of this module (which is great btw) and havent found any links to `CoreBluetooth` or other APIs different than `CoreLocation` that does not need that permission because the only thing it cares about is location permissions.

Have you had any issues with submitting the app so far?

frostney, 2015-11-24:
Let me dig into this. There may have been some changes with iOS 9 I'm not aware of. I've run into something similar before. In an app we submitted we used background modes, but didn't actually require background modes in the app itself.

Could you provide some additional information if possible: Which iOS version are you targeting? Xcode version? Are you using background modes?

zackify, 2015-11-24:
:+1: thanks a ton for this library, I'm hoping this gets resolved before I submit my app.

grabbou, 2015-11-24:
Hey, thanks for the answer! 

- XCode 7.1, 
- Base SDK: iOS 9.1
- Target version: 9.0

I don't use any other background modes, just built-in location from `react-native`. I am using `location` in the background mode to receive constant location updates since the application is highly coords oriented. But I might be misconfiguring that.

Another case (rather impossible) is that I turn on the beacons module after user selects a venue that supports that. So there might be the case that Apple haven't seen the location popup and haven't set up any beacons nearby. But on the other hand, I'd expect them to search for references in native files first and I think that's what they usually do.

frostney, 2015-11-24:
It turns out there have been some changes in iOS 9 relating to the background functionality. Specifically the `allowsBackgroundLocationUpdates` of the location manager needs to be set to `true`/`YES`, otherwise it's not listening to background updates which may be the reason of the rejection.

I'll fix that up as soon as I can, but I may not be able to release a new version before tomorrow.

grabbou, 2015-11-24:
No worries, thanks for the feedback. I'll try to check it out in my project and if it works, I'll get back to you with more details.

MossP, 2015-12-08:
Did this get resolved? Just looking at possibly using this for something.

frostney, 2015-12-08:
I'm going to push out a new release tonight with the fix I described above. I'm sorry I couldn't get to it earlier.

MossP, 2015-12-08:
That's great news @frostney

grabbou, 2015-12-08:
I've basically removed that flag as per Apple fix and still waiting for the results (it works anyway)

rformato, 2017-02-14:
As far as I know, to use region monitoring (not ranging) with beacons on iOS you don't need any background mode, but just be sure that the location authorization is set to "Always". Region Monitoring enter event works even when the app is terminated, relauncing it for a short amount of time so that you can notify the user. If you need ranging in background, then you need the location update background mode.

You can read more on https://developer.apple.com/reference/corelocation/cllocationmanager

If you don't have a use case for background location updates, better not to enable it, both to save battery and time in case apple asks why you need it.

johnlev, 2017-04-26:
Anyone know if this has been resolved? Just got a number of very insistent messages from the Resolution Center that I don't need background location modes. During a long phone call with the people at Apple they tried to convince me that to detect enter and exit events I don't need background location, and yet trying it without causes monitoring to fail.