frostney/react-native-ibeacon #23: transmitting as an ibeacon

sergethompson, 2016-01-14:
It would be great if you could switch between a receiver and a transmitter using this native code. Is a transmitting feature coming in a future push?

zackify, 2016-01-14:
How would this be practical though? There is a demo app you can get from
apple's website that will let your phone act as a beacon.

frostney, 2016-01-14:
@sergethompson Would you mind explaining your use case in a bit more detail?

sergethompson, 2016-01-14:
I think it would be interesting in the case of doing something like taking
attendance where one teacher(receiver) can see if all students(beacons) are
there using a singular app. This would all have to be verified over a
server with the specific individual uuids. I can help build it if you lead
me in the right direction. I just have no idea where to start.

sergethompson, 2016-01-14:
> *c. 603.502.0290 <603.502.0290>*

I think it would be interesting in the case of doing something like taking
attendance where one teacher(receiver) can see if all students(beacons) are
there using a singular app. This would all have to be verified over a
server with the specific individual uuids. I can help build it if you lead
me in the right direction. I just have no idea where to start.

zackify, 2016-01-14:
The correct way to do this would be to have a beacon in the class room, and
when the beacon is in range, do a request to your server with the user's
info. That way you can let teachers login and see who is in range.

sergethompson, 2016-01-14:
Thanks for the feedback. I also thought it would be cool to have a mobile solution that could be used on field trips too so I don't have to worry about always having the beacon and be able to check status all within the same app.

zackify, 2016-01-14:
That defeats the point of a beacon at that point. It would be better to
track their gps location in that case. If you ask me that's also too much
of a privacy concern haha

marktyers, 2019-12-08:
Since this uses the same Apple framework as detecting iBeacons it would be a really good idea to implement. Happy to help.