frostney/react-native-ibeacon #26: Bluetooth permission

albo1337, 2016-03-29:
Hello team,

I know that it's not possible to programmatically turn on bluetooth. But how can I display a popup window that the user can open the bluetooth settings, if the user hasn't turned bluetooth on.

Best regards

frostney, 2016-03-29:
Hi @albo1337, I ran across the same issue when I was using this module in production. For that reason, I made a separate module called [`react-native-bluetooth-state`](https://github.com/frostney/react-native-bluetooth-state) that checks if Bluetooth is active.
Alternatively you could make a native extension for that yourself, something similar to http://stackoverflow.com/a/23838229/1544387. (`react-native-bluetooth-state` is using that under the hood.)

Feel free to re-open the issue if that doesn't answer your question.

closed 2016-03-29 by frostney

albo1337, 2016-03-30:
Hi @frostney thank you very much for your answer. I'm using your module. My problem is, that I have no idea how to implement the listener. Could you give me a simple example that pop up a window, when bluetooth is off?

Would be great!