frostney/react-native-ibeacon #28: React-Native-ibeacon for android soon release

albo1337, 2016-04-02:
Hello Team,

I will post soon (I think this week) a simple method to reach the beacons on android devices using react native. It's just a wrapper of the estimote SDK, but I think it can help you all for now.

Best regards

Albo

mmazzarolo, 2016-04-04:
Hello @albo1337 , does the receiver work in the background too? (Using something like a service/broadcast-receiver). 

I've planed to add a background service on a tweaked version of [react-native-alt-beacon](https://github.com/octavioturra/react-native-alt-beacon), but maybe waiting for your implementation would be an easier solution 🎱 .

albo1337, 2016-04-13:
Hello Team,

here we go: https://github.com/ti8m/react-native-ibeacon-android

I've not tested it well, because I hadn't that much time. But there is an example App within which you can use to test a little bit.

mmazzarolo, 2016-04-13:
Nice @albo1337 ! Well... [it seems we've been trying to do almost the same thing](https://github.com/mmazzarolo/react-native-beacons-android) 😄   
(Still need to complete the docs of mine though)

albo1337, 2016-04-13:
Nice @mmazzarolo ! Maybe we can publish a complete react-native-ibeacon version for both OS.

frostney, 2016-04-13:
This is amazing :tada: 
Thank you. At the very least I'll add a link in the readme, as for integrating that into this repository I would be totally be up for that after some thorough testing.