frostney/react-native-ibeacon #29: Beacon-detection without location permission

albo1337, 2016-05-12:
Is it possible to detect Beacons without location permission? I think we just need bluetooth to detect beacons and not location permission.

Kind regards,
albo1337

frostney, 2016-05-12:
The problem here is that the iBeacon functionality in the iOS SDK is bundled with `CoreLocation`. `CoreLocation` needs location permission, in order to use any of its API methods.

If you were to disable location permissions, you would not be able to get the location updates (ranging, scanning) for the beacons.

closed 2016-05-12 by frostney

albo1337, 2016-05-12:
Damn! Ok, thanks a lot for that quick answer.

albo1337, 2016-05-12:
Then I have another problem which is similar to this one (this time to check for location):

https://github.com/frostney/react-native-bluetooth-state/issues/3