frostney/react-native-ibeacon #30: RN update 0.26 breaks `Beacons.getAuthorizationStatus`

Andruschenko, 2016-05-24:
After updating to react native version 0.26, the calling `Beacons.getAuthorizationStatus` returns `'notDetermined'`.

It seems like the new react native update includes some breaking change for this module.

### Version
0.6.0

### Steps to reproduce
1.  Update react native to version 0.26
2.  Call `Beacons.requestWhenInUseAuthorization();`
3.  

### Expected behavior
Authorization granted

### Actual behavior
Authorization is not granted and beacon ranging does not begin.

frostney, 2016-05-24:
Thank you for reporting the issue. I'll get to it as soon as I can, but it might take me a few days. Feel free to leave suggestions if you find something about why this issue happens.

Andruschenko, 2016-05-24:
False alarm! I accidentally removed `<key>NSLocationWhenInUseUsageDescription</key>` in the `info.plist` file in another commit which caused the issue. 
However, thanks for replying that fast @frostney! You may close the issue. Sorry for the confusion. At least I can hereby confirm compatibility with RN 0.26.

btw, thanks so much for this great module!

closed 2016-05-24 by Andruschenko