frostney/react-native-ibeacon #35: Conflict with react-native-background-geolocation plugin, possibly because plugin assumes existence of proximityUUID parameter

skokhuis, 2016-07-04:
Hi,

This plugin conflicts with https://github.com/transistorsoft/react-native-background-geolocation. The author of this geolocation plugin suggests this plugin is responding to “his” geofences, interrogating it for the proximityUUID key which doesn't exist.

Could this be resolved by checking for the existence of the proximityUUID parameter? I have too little C knowledge to fix it myself. Thanks! Also for this great plugin.

This is the result with both plugins in a project when a geofence is crossed:

> 2016-07-04 14:16:49.933 GeoFence[2772:401504] -[CLCircularRegion proximityUUID]: unrecognized selector sent to instance 0x1278f0800
> 
> 2016-07-04 14:16:49.936 GeoFence[2772:401504] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[CLCircularRegion proximityUUID]: unrecognized selector sent to instance 0x1278f0800'

skokhuis, 2016-07-04:
Adding a check if the proximityUUID property exists seems to have fixed it. File RNBeacon.m, updated last part to:

```
-(void)locationManager:(CLLocationManager *)manager
        didEnterRegion:(CLBeaconRegion *)region {
    if (! [region respondsToSelector:@selector(proximityUUID)]) {
        return;
    }

    NSDictionary *event = @{
                            @"region": region.identifier,
                            @"uuid": [region.proximityUUID UUIDString],
                            };

    [self.bridge.eventDispatcher sendDeviceEventWithName:@"regionDidEnter" body:event];
}

-(void)locationManager:(CLLocationManager *)manager
         didExitRegion:(CLBeaconRegion *)region {
    if (! [region respondsToSelector:@selector(proximityUUID)]) {
        return;
    }
    NSDictionary *event = @{
                            @"region": region.identifier,
                            @"uuid": [region.proximityUUID UUIDString],
                            };
    
    [self.bridge.eventDispatcher sendDeviceEventWithName:@"regionDidExit" body:event];
}

```

ca057, 2017-03-06:
Thanks for pointing out the fix. I had a similar issue and implemented your solution in a fork which you can check out [here](https://github.com/Artirigo/react-native-ibeacon/pull/1). Does anyone know if there should be more checks added? Otherwise I would test it and eventually file a PR here in the base repo.

plbrault, 2017-03-17:
Thank you @skokhuis and @ca057 for the fix. I had the exact same issue in my project.