frostney/react-native-ibeacon #40: regionDidEnter/regionDidExit are not firing

montehansen, 2016-12-06:
### Version
0.6.0
react-native-cli: 1.2.0
react-native: 0.38.0

### Steps to reproduce
```
DeviceEventEmitter.addListener ( "regionDidEnter", (region) =>
{
  console.log( "***** regionDidEnter: " + JSON.stringify(region || {}, null, "\t") );
});

DeviceEventEmitter.addListener ( "regionDidExit", (region) =>
{
  console.log( "***** regionDidExit: " + JSON.stringify(region || {}, null, "\t") );
});

Beacons.requestAlwaysAuthorization();
// Beacons.requestWhenInUseAuthorization();
Beacons.startMonitoringForRegion(Entrance.REGION);
Beacons.startRangingBeaconsInRegion(Entrance.REGION);
Beacons.startUpdatingLocation();
```
### Expected behavior
regionDidEnter/regionDidExit should fire.

### Actual behavior
regionDidEnter does not to fire, or rarely does. I am using Estimote and Kontakt.IO beacons and have the same experience for both. For the Estimote, I did test using the ranging method, and the beaconsDidRange event does appear to fire and I do see change in the proximity field. 

Any thoughts?

Thanks,

Monte

montehansen, 2016-12-06:
Hmmm. I moved to a different location and using the same beacons have been able to get the regionXxx methods to fire. Strange. I will close but would love some insight from anyone on that regarding experiences of what I assume could be interference?

devand123, 2017-09-27:
I'm experiencing the same. On latest version of RN