frostney/react-native-ibeacon #46: Background mode doesn't work !

Clowning, 2017-02-04:
Hey,

I'm working on an app which send a notification to user when a ibeacon is around him ! But i have a problem, when my app is running on screen, everything works but when i go on home, nothing ! The application do anything ! 

I've enabled Background Mode, add some key in info.plist ! I've tried to add some line : 

```
 if ([launchOptions objectForKey:@"UIApplicationLaunchOptionsLocationKey"]) {
    NSLog(@"DID RECIEVE REGION");
}
```

But it's not fired ! Do you have an idea?

My react-native version : 0.40
IOS : 10.2

Thank you!

shodanuk, 2017-02-06:
I'm having the exact same problem. I believe that the javascript is running ok in the background as I'm seeing logs from regionDidEnter & regionDidExit event handlers but I don't believe didFinishLaunchingWithOptions is firing at all

shodanuk, 2017-02-08:
I got this working... eventually.

It appears that didFinishLaunchingWithOptions does not fire when a regionDidEnter event is triggered when the app is in the background. I tried this with a pure native iOS app and it displayed the same behaviour, so I don't think it has anything to do with React Native or this library.

As I mentioned in my last comment, regionDidEnter & regionDidExit continue to fire when the app is in the background so it's quite possible to listen for these events and trigger a location notification from your Javascript code. 

Check out [react-native-push-notification ](https://github.com/zo0r/react-native-push-notification) for a cross platform notification lib.

nikolakanacki, 2017-07-12:
@shodanuk how exactly did you get any of this to work? I've had no luck with it