frostney/react-native-ibeacon #51: duplicate interface definition for class 'RTBridge'

Dri4n, 2017-03-29:
### Version
0.6.0

### Steps to reproduce

1.  Install react-native-ibeacons
2.  react-native link & link manually librarary
3.  Semantic issues

[image]

react-native versión is 

[image]

### Actual behavior
any ideas for this ?

rem7, 2017-04-11:
@Dri4n did you figure this out?

srikanthkyatham, 2017-04-18:
I solved it by commenting 

//#import "RCTBridge.h" 

in RNBeacon.m 

But the issue every new yarn install or npm install would result in the same issue all over again.
Need to submit a PR