frostney/react-native-ibeacon #58: installation issue

nemr, 2018-04-24:
can you help me out with setting up this module ? I tired dragging the project file into my project folder , tried adding it to the library in xcode but they both didnt work 

### Steps to reproduce 1st attempt
1.  yarn add react-native-ibeacon
2. I Dragged (Moved)  RNBeacon.xcodeproj from node_modules/react-native-ibeacon to my project direcotory folder 
[image]

3.  add libRNBeacon.a and CoreLocation.framework on build phases 

### Actual behavior
I get the following error : 
clang: error: no such file or directory: '/Users/nero/Desktop/bleBeacon/ios/RNBeacon.m'
clang: error: no input files

### Steps to reproduce 2nd attempt 
1.  yarn add react-native-ibeacon
2. I Dragged  RNBeacon.xcodeproj from node_modules/react-native-ibeacon to the library in the xcode project :
[image]

3.  add libRNBeacon.a and CoreLocation.framework on build phases 

### Actual behavior
I get the following errors:
[image]

amsul, 2018-06-15:
@nemr I'm having the same issue. Were you able to figure a way around?

pushpendraKh, 2018-06-28:
@amsul  @nemr is this issue solved?